ESP32-WiPhone/wiphone-firmware #2: No support for SIP/UDP

zakx, 2021-03-24:
Apparently, the project does not support SIP over UDP. Is this intentional?
Somehow all my three SIP providers only offer SIP over UDP; one offers SIP over TCP+TLS (Port 5061). Therefore, at the moment, I cannot use the SIP functionality.

olivierBautista, 2021-04-09:
Same issu, in France, OVH my SIP providers is only SIP over UDP

K1rdro, 2021-04-10:
I'm pretty sure mine works using 5060 UDP:
[image]
The above is with VOIP.ms. I may be completely wrong though, I haven't tested SIP to SIP. It took a bit to get my account configured, and finding a decent provider in US with a SIP Trunk was a pain, since I'll be doing little to no SIP to SIP.

zakx, 2021-04-10:
No, there is no code supporting SIP over UDP in the firmware.

jhwegener, 2021-04-22:
Are there any plans to integrate this SIP over UDP?

MrJH, 2021-04-22:
Are there any plans to integrate support for SIP over UDP?

nosliwneb, 2021-04-22:
At the moment we are prioritizing getting the hardware shipped out,
documentation, and critical software issues.

Once that's done, we will collect all the less critical bugs and features
we/everyone else want/s and prioritize them.

SIP over UDP should be pretty high on the feature list.