ESP32-WiPhone/wiphone-firmware #3: Audio Settings Not Applying Correctly

K1rdro, 2021-04-07:
When using Audio Settings, Loudspeaker Volume and Ear Speaker Volume seem to both only work on the loudspeaker. 

When Loudspeaker is turned up all the way, and Ear Speaker is turned all the way down, Loudspeaker is at max volume and Ear Speaker is off (Which is expected).

When Loudspeaker is turned all the way down, and Ear Speaker is turned all the way up, Loudspeaker is at max volume and Ear Speaker is off (unexpected).

Ear Speaker remains off regardless of Audio Setting level for it (unexpected). 

When in a call however, it appears that Ear Speaker can be manipulated using the up and down arrows, however, all audio devices match volume and respond, so it's hard to tell.

xopr, 2022-08-14:
Tested on version 0.8.30: I noticed when dialing out, the left button says: "Loud Spkr", but it already starts with loud speaker.
After pressing once, it changes to "Ear Spkr", but the audio doesn't change.
After pressing again, it changes to "Loud Spkr", and the audio changes to "Ear Spkr" (as expected)

nosliwneb, 2022-08-14:
Call audio was previously stuck coming out the loudspeaker only. In the
0.8.30 update we decided to add the audio adjustment even though the UI
still isn't 100% working right. We are going to tweak it more in the next
release.

xopr, 2022-08-18:
I think the fix for the initial out-of-sync state might actually be easy (but I currently can't test):
In the constructor of `CallApp` in `GUI.cpp`, the last instruction should probably be `audio->chooseSpeaker(EARSPEAKER);`