ESP32-WiPhone/wiphone-firmware #14: When I make a fail call, the firmware hangs and ends in reboot.

tunyx, 2022-01-17:
When I make a failed call, the firmware hangs and ends in reboot.

gigawatts, 2022-02-08:
I can confirm this behavior as well.

gigawatts, 2022-07-13:
Testing out FW v0.8.29rc2, connected to a LAN Asterisk server via UDP, I can make calls from the WiPhone, but receiving calls is very sketchy. I am using a Cisco IP Phone as the other line to test with. Answering an inbound call on the WiPhone now almost always results in being stuck at the "Accepting" state (and no audio is passed) despite the fact that the Asterisk logs state it responded with "Answered". Sometimes it answers, but then immediately falls back to the Home screen, and other times it just crashes and reboots when answering. Answering seemed to work better in previous firmware versions.

I will switch to TCP next to see if that makes any difference in behavior, but generally use UDP as that is what my Cisco Phones support.