ESP32-WiPhone/wiphone-firmware #18: Echo in calls

fortalloy, 2022-04-08:
call recipient at long distance (>5000 km away) tell me there is an echo in the call.  This echo does not happen using the same voip service from a computer or smartphone app, nor is reported calls placed to recipients in nearer locations, so a common denominator seems to be the handset (the WiPhone).  Not sure if half duplex audio would fix it, but it's a possibility.  The audio incoming and audio out might need to be separated somehow.  I have a suspicion that the audio in just plays back in the audio out, so the 2X travel time is long enough to exhibit this echo, since it only happens for long distance and not short.  But that is just a suspicion, the quality of the switching hardware at the other end of the call can also be the issue.

Loria, 2022-06-07:
Some thoughts about that:
- Is the connection (with echo) a direct connection (WiPhone <-> WiPhone (without any services inbetween)? In this case the actual echo delay might be of interest then.
- Is there an echo canceling done within the WiPhone firmware ?
(sorry in case this is already common knowledge)

fortalloy, 2022-06-08:
1-No, not WiPhone 2 Wiphone call.  The issue happens between WiPhone to landline or WiPhone to telco cell, the landline or cell user complains of echo.  WiPhone does not have echo (I am calling relatives living at a distance from my WiPhone and the are using more conventional telco landline or cell handsets.)

2- I do not know the answer; you will have to discuss to WiPhone developers ‘in the know.’

nosliwneb, 2022-06-11:
We don't currently have echo cancellation or packet loss concealment.