ESP32-WiPhone/wiphone-firmware #19: weird interaction with freeswitch server

brianmay, 2022-04-11:
Latest testing firmware.

Incoming phone calls:

* Sometimes doesn't ring. With caller hanging.
* Other times will ring, but when trying to answer, it hangs.
* If caller tries to abort phone call while ringing, the phone continues to ring.

Outgoing phone calls:

* If try to abort outgoing call, remote phone continues ringing. I think call will get transferred to voice mail if not answered.
* If call is answered, then everything works perfectly.

brianmay, 2022-04-12:
If I "reject" incoming call by pushing the hangup button and then the "reject" button (need to test again to get precise sequence), phone stops ringing, the display goes back to the home screen, but the phone continue vibrating. Eventually the ring screen shows again and I can reject it properly.

Loria, 2022-06-07:
> Outgoing phone calls:
> 
>     * If try to abort outgoing call, remote phone continues ringing. I think call will get transferred to voice mail if not answered.
> 
>     * If call is answered, then everything works perfectly.

Same happens if using a "fritz!box" as a SIP Gateway. It looks like, if I abort the call, the operation actually is not aborted downstream ..

brianmay, 2022-08-01:
New update. There was a config error in one of my clients, was unexpectedly using a STUN server when I didn't not expect or want this to be the case. That might have caused a bit of confusion.

Now outgoing calls seem to work perfectly (fingers crossed...). But there are problems with incoming calls, don't be able to accept them.

This is version 0.8.29rc2, which I believe is the latest version.

nosliwneb, 2022-08-06:
@brianmay can you share what type of SIP server you are interacting with,
or possibly give us a login on it so we can check what's happening?

@Loria, do you have the same issue if you upgrade to firmware version
0.8.29 (0.8.29rc2 is the same thing, if you are on the development URL). We
made quite a few improvements to the SIP signaling in the latest release,
so it's not surprising if you are seeing this sort of thing in 0.8.27.
Message ID: ***@***.***>

brianmay, 2022-08-07:
Freeswitch

brianmay, 2022-08-09:
Unfortunately, having some weird problems with external communications with my freeswitch right now, it probably won't help giving you access to it (unless you want to help debug unrelated issues...).

(sidenote: when the phone rings, the screen doesn't turn on, this is a bit disconcerting...)

Note: Right now aborting the call appears to be working, strange, I don't think that was always the case.

Here is a GIST showing the full freeswitch logs, with SIP tracing, showing the failure: https://gist.github.com/brianmay/8a20dcdab275838192334513eb2a72fe

From 1004 I called 1003, let it ring, tried to answer it (nothing happens), then cancel the call from 1001 (it reverts to voicemail).

To my untrained eye it looks like 1001 is clearly telling freeswitch it wants to answer the call, and freeswitch is receiving this, but for some reason, freeswitch is ignoring it.

Lines 373-393:

```
recv 635 bytes from tcp/[192.168.5.213]:62143 at 08:19:39.161251:
------------------------------------------------------------------------
SIP/2.0 200 OK
Via: SIP/2.0/TCP 59.167.180.194;branch=z9hG4bKyFtKy58BjFe3j
To: "home" <sip:1001@sip.microcomaustralia.com.au>;tag=ztwF11YhW
From: "Brian May" <sip:1004@sip.microcomaustralia.com.au>;tag=yvy7Xtvt7Dvta
Call-ID: 3176ae4d-92d4-123b-138b-00e04c006895
CSeq: 44540 INVITE
Contact: <sip:18446288@192.168.5.213:62143;transport=tcp;ob>;+sip.instance="<urn:uuid:b5fc7dec-40e2-11e9-b210-98f4ab6cb238>"
Content-Type: application/sdp
Content-Length: 168

v=0
o=- 3752000721 3752000721 IN IP4 192.168.5.213
s=WiPhone
t=0 0
m=audio 54002 RTP/AVP 0
c=IN IP4 192.168.5.213
a=rtcp:54003
a=rtpmap:0 PCMU/8000
a=sendrecv
```

nosliwneb, 2022-08-11:
@brianmay would it be possible to share these config files?

directory/default/sip_usr*.xml
sip_profile/internal.xml
dialplan/

brianmay, 2022-08-12:
Is there somehow I can send you this privately? Don't feel particularly comfortable letting the entire Internet have access to my config files...

nosliwneb, 2022-08-12:
sure. please email them to ***@***.***
Message ID: ***@***.***>

>

brianmay, 2022-08-12:
That email address didn't come through :-(

nosliwneb, 2022-08-12:
support
at
wiphone.io

brianmay, 2022-08-12:
Got it, thanks

brianmay, 2022-08-12:
Sent you the config.

brianmay, 2022-08-19:
Update: With test software I was provided with, incoming calls now work fine, but now random intermittent problem with outgoing calls.