ESP32-WiPhone/wiphone-firmware #22: Won't connect to already configured WIFI Networks.

Loria, 2022-06-08:
If move the device from one Wifi Network to another it won't connect automaticaly to the other already known Network. I have to manually Scan and select the network to connect .. 
But this might be an issue with ESP32IDIF..

nosliwneb, 2022-06-08:
Currently we only have a "primary" network. This means it will save
credentials for multiple networks, but it will only try to automatically
connect to one.

This could definitely be improved. At the moment we are still working
through SIP and making calling reliable.

In our testing, scanning for networks will drain the battery very quickly,
so whatever algorithm we use will need to be designed with battery life in
mind.

Loria, 2022-06-08:
Ok, this does explain. I thought, it could be possible to move the network connection into a thread, which scans and connects to "allowed" networks.