ESP32-WiPhone/wiphone-firmware #29: Feature request: Right arrow to enter submenu

xopr, 2022-08-14:
Menu forward and backward navigation by using _right_ arrow key.
Similar as in **Phonebook** and **View contact**, it would be handy to thumb navigate using up/down for item selection, and (left/)right for (leaving/)entering a submenu.
Maybe even in every case except where the left/right keys are in use (textbox navigation and in-app use)

This would at least be very handy in the `action=GUI_ACTION_SUBMENU`
Maybe _left_ arrow could be used as well to navigate back.

I might circle back to this improvement if I can figure out some details.

nosliwneb, 2022-08-14:
The arrow pad has a center button which can be used to enter menu items.

As long as left/right aren't already in use for something else inside other
menus, using them for forward/back makes sense to me.

xopr, 2022-08-14:
Yes, agreed: this feature might be considered "with care": I assume everything that is not a submenu will break if this would be implemented too rigorously (not sure if it is even possible to differentiate).

I just noticed that my habit was to navigate "right" to jump into a submenu (currently, I prefer using the left softkey to make sure I don't accidentally press an arrow key when selecting a menu item (I probably have clumsy fingers; see my quick fix subnote in #28 :)

It might take a week for me to look into this so if anyone wants to look into it: I won't feel caught up.