ESP32-WiPhone/wiphone-firmware #30: Enable caret cursor in dial app

xopr, 2022-08-14:
When the user starts to dial (`ENTER_DIAL_APP` -> `GUI_APP_DIALING`), there is no visual indication where the cursor is.

This might especially be confusing after the first `WIPHONE_KEY_LEFT` event where a user can insert or delete digits in the middle of the dialed number; caret enabling after the first event or when the position is not at the very end might be least intrusive solution.

nosliwneb, 2022-08-14:
you can't see that black cursor on the black background? :)

(I'm guessing, but 90% sure that's what's happening)

yeah, we do need to add (or show) the cursor
Message ID: ***@***.***>
>
>

xopr, 2022-08-23:
@nosliwneb you're right, I had a look at the input widgets and they all show a (fixed color) caret.
I _think_ [this patch](https://github.com/ACKspace/WiPhone_build/blob/master/patches/caret.patch) should fix it, but I'm currently unable to test it.
(let me know if you have problems with a public patch and I'll make the repo private for the time being)