ESP32-WiPhone/wiphone-firmware #33: jmp.chat SIP account issues

Torbuntu, 2023-01-19:
I've been trying to get my jmp.chat SIP credentials to work with the WiPhone. So far I've ran into a couple of smaller issues. 

#### port 5060 -> 5008 for signing into account:
There are a lot of hard coded ports set to 5060. This seems to be a commonly used port for SIP so it makes sense, however for jmp.chat it is expecting 5008. By manually replacing all of the 5060 -> 5008 I've been able to get my account to sign in successfully :tada: 

#### SIP Syntax error when attempting to place call
This one was weird to debug and took quite a bit of reading of the code and placing loggers all over the place. Finally it came down to the buffer in `sdpBody` not being large enough. By brute setting the buffer length to 3x the format instead of 2x it was able to resolve the 400 bad gateway SIP Syntax Errors I was getting.
 `char buff[3*strlen(format)];`

#### 500 Internal Server Errors
Now I'm not sure where to go from here. When placing a call I can see it tries to communicate but resolves into a 500. I tried tinkering with the Contact: line as I think using the randomly generated number for phoneNumber may be an issue if it expects a real usable value there. But maybe that's the wrong path. The method setting 

I can provide logs and tweaks that can be helpful. I'll continue to poke at it otherwise as I think it is very close to working!

ossguy, 2023-01-19:
On the port number, the WiPhone should always be using DNS SRV records for SIP first, and only if those don't exist trying other ports (like 5060 or similar).  This would solve the situation for JMP SIP accounts, for example, and is generally the standard way to do things.

Torbuntu, 2023-06-13:
I've still been poking at this but I've made basically net negative progress with all my destructive tinkering :D