ESP32-WiPhone/wiphone-firmware #40: 0.8.32rc1 to 0.8.32rc2 update failure

gigawatts, 2023-07-31:
Current branch: testing
Current firmware: 0.8.32rc1

The testing branch shows an update (0.8.32rc2), but choosing install seems to just initiate a reboot, as it never downloads and applies rc2.

nosliwneb, 2023-07-31:
The firmware update screen should show an error message. Can you share that?

Sometimes if the update fails we disable automatic updates to prevent
bootlooping but if you navigate to the firmware update screen you can still
manually initiate an update. Is that what you are currently doing?

Could you move to a different network and give it another try? We've just
done a test on our side and the firmware server seems to be working as
expected.
Message ID: ***@***.***

gigawatts, 2023-07-31:
I have auto updates off, and always do a manual check and install, so yes,
that's what I did this time too. No error message, just the reboot. Does
there serial terminal spit out any messages during the update process? I
didn't see anything readable on 9600 or 115000 baud (other than the ESP32
bootloader messages during power on), haven't tried the other common baud
rates yet.

I can give it another try with a different network later. One thing to
note, I did just flash a completely different firmware (meshtastic) to the
WiPhone, then restored a backup of the flash I had taken previously, so as
not to loose my settings, so it's possible that has complicated this update
somehow.

One feature request that would be useful in this situation is the ability
to export and restore configs (SIP server, contacts, call logs, etc)
to/from an SD card. Then I could do a full flash erase, flash the latest
firmware, and restore my configs.