ESP32-WiPhone/wiphone-firmware #41: Audio recorder doesn't save or play back any audio

GIGObug, 2023-08-01:
I have the firmware updated, and a 64 Gb microSD card installed. Start the audio recorder, record a 5 second audio and press the save button. Message appears "unable to save audio".

nosliwneb, 2023-08-01:
It will only work with an SD card that supports SPI mode. Many larger cards
(over 16GB or so) do not have good SPI mode support.

Please try a smaller card and see if that helps.

roberto-morado, 2023-08-02:
I can testify to what @nosliwneb is saying, I had the same problem with a 32gb, I changed to an 8gb and the problem went away 👍

GIGObug, 2023-08-02:
Thank you for your replies. It makes sense, however what do I have to look for on these microSD cards? I put a class 10, 8Gb card in, and it still doesn't work. I've never heard of SPI mode.

GIGObug, 2023-08-02:
Thank you. What should I be looking for? I only have class 10 cards. I tried a 4Gb card yesterday and still nothing. Should it be formatted to a particular file system? I can’t seem to find any information on the WiPhone pages or YouTube videos on sd cards. Lutz