ESP32-WiPhone/wiphone-firmware #43: Never got source code

Nikguy321, 2023-08-15:
I was supposed to get the source code with the phone but never did. In fact, I never got a tracking email, but I did get the phone.... Could I get a copy of the source code?

morganrallen, 2023-08-27:
Right, is there a plan/timeline for getting this repo up and running? The projects tagline is..
> WiPhone: An Open Source Phone That's Really Yours
but after 9 months since I've received mine there has been no movement on opening the project up. I would love to contribute to this project, specifically around LoRa (I help maintain arduino-LoRa and an esp32 specific sx1276 driver (git.oit.cloud/morgan/esp32-lora)) but having to email and ask for source updates isn't a viable way to give back.

Nikguy321, 2023-08-28:
> Right, is there a plan/timeline for getting this repo up and running? The projects tagline is..
> 
> > WiPhone: An Open Source Phone That's Really Yours
> 
> but after 9 months since I've received mine there has been no movement on opening the project up. I would love to contribute to this project, specifically around LoRa (I help maintain arduino-LoRa and an esp32 specific sx1276 driver (https://git.oit.cloud/morgan/esp32-lora)) but having to email and ask for source updates isn't a viable way to give back.

Hey, I got the firmware (well one version behind). I can share it with you if that is allowed. I was able to get it to compile combined with the "core files" link on their "programing instructions" page on their site.

Nikguy321, 2023-08-28:
> I've also requested the latest, but I think the broader point still stands, phones have been shipping for well over a year, it's time to put the Open in Open Source. I'm getting to the point I'm considering writing a whole new OS from scratch, which I really don't want to do.

Well if you do rewrite the OS, I think you'd be single handedly reviving this project.   It would be great if it used a lorawan protocol in such a way that you could make a meshtastic app for it while using the lora daughter board.   I know like 80% of the ppl who bought this were under the impression that they could make a meshtastic app for it.  Maybe you can, but I'm not skilled enough to do that... 

Let me know if you want the source code to base whatever you are doing on

morganrallen, 2023-08-28:
Personally I wouldn't use either LoRaWAN or Meshtastic, they both share the same issue as they're designed as complete monolithic projects. You flash Meshtastic onto your device and use it in it's entirety. This basically means you cannot use it along side an existing project, like an already large phone OS. LoRaWAN is slightly better but shares the same issues. Which is a shame, because conceptually they're both great projects, but the all or nothing approach to writing software just doesn't work out if trying to integrate into something else.

>  I know like 80% of the ppl who bought this were under the impression that they could make a meshtastic app for it

Not sure where they picked that idea up but from what I understand, they'd be better off approaching Meshtastic to add WiPhone support instead of the other way around.

Nikguy321, 2023-08-28:
Yeah I agree mostly, though I do use meshtastic a lot when out camping and so on. 

But early on the wiphone development, meshtastic reached out and wanted to merge their firmware with the wiphone firmware in a way that it would just be a separate app. So you'd retain your wiphone GUI, and you would passively receive messages, and then launch the app to see the conversation and to send messages.  I guess wiphone team didn't want to go that route I guess? 

But you can get the meshtastic source readily, and some of the meshtastic developers said it wouldn't be the worst project in the world to merge the firmware like I mentioned above, but I think they just have higher priorities.  I've been trying to learn more so I can do it, but man, I'm a long ways away.

kenjreno, 2023-12-20:
Last version I was able to get was 8.27. I see 8.32 is currently loaded on the 2 that I have. Does anyone have more current version of the source than what I have?

theSoberSobber, 2024-02-05:
> Last version I was able to get was 8.27. I see 8.32 is currently loaded on the 2 that I have. Does anyone have more current version of the source than what I have?

How'd you get it?

ebraminio, 2024-07-19:
It is available here https://github.com/playfultechnology/wiphone apparently.

I totally understand the good willing of ESP32-WiPhone developers, even I guess the plan was to release the source to public from the beginning they wanted to polish it enough before first but never felt are there, they perhaps should've embraced opensource motto, https://en.wikipedia.org/wiki/Release_early,_release_often

brianmay, 2024-07-26:
That code looks like it is old.

With the lack of any official word for months now, I can't help but feel this project is basically dead. And sure, real life problems can get in the way and cause this. No drama.

But if this is the case, it is more important to publish the up to date source code, so that others can take over. Please do so.

gigawatts, 2024-07-26:
The most up-to-date (official) copy of the source I can find is here: https://wiphone.io/static/releases/firmware/WiPhone-0.8.30.zip

brianmay, 2024-07-26:
So nothing newer then 0.8.30?

According to above messages version 0.8.32 was the latest as of December last yet.