ESP32-WiPhone/wiphone-firmware #44: Any updates? What can the community do to help you all get back in fall development mode?

Nikguy321, 2023-10-15:
I'm not great at coding, but I'd like to know what we can do as a community to keep this project going. I don't want it to die down. Let us know, and I can try to track down willing developers.

Mildmanneredmiles, 2023-10-27:
I agree. I am very interested in this project and feel like great things could come of this. However, I have very little coding experience. I would be happy to help in any way to get this project moving forward again. Would love to see some wort of LoRa Mesh integration.

gdan342, 2023-11-11:
I was an early backer, and haven't seen anything happen with this device for 2 years. It's very disappointing. It has so much potential, but until we have a source to fork off of and make our own projects it's absolutely worthless. Could have been the coolest device out there

Nikguy321, 2023-11-11:
> I was an early backer, and haven't seen anything happen with this device for 2 years. It's very disappointing. It has so much potential, but until we have a source to fork off of and make our own projects it's absolutely worthless. Could have been the coolest device out there

Dm me, I have the source, well a version-ago's source code. I'm just not too great at coding unfortunately, so I can't do much magic with it...

gdan342, 2023-11-11:
> > I was an early backer, and haven't seen anything happen with this device for 2 years. It's very disappointing. It has so much potential, but until we have a source to fork off of and make our own projects it's absolutely worthless. Could have been the coolest device out there
> 
> Dm me, I have the source, well a version-ago's source code. I'm just not too great at coding unfortunately, so I can't do much magic with it...
> 
> Edit, I'll just post the link here. Why not.
> 
> https://drive.google.com/drive/folders/1dlPjPe960pd_3zQCrj8OW05oHb8o0nCY
> 
> Check here and see the Readme and the "look here" pic. Hope links like this are ok

> > I was an early backer, and haven't seen anything happen with this device for 2 years. It's very disappointing. It has so much potential, but until we have a source to fork off of and make our own projects it's absolutely worthless. Could have been the coolest device out there
> 
> Dm me, I have the source, well a version-ago's source code. I'm just not too great at coding unfortunately, so I can't do much magic with it...
> 
> Edit, I'll just post the link here. Why not.
> 
> https://drive.google.com/drive/folders/1dlPjPe960pd_3zQCrj8OW05oHb8o0nCY
> 
> Check here and see the Readme and the "look here" pic. Hope links like this are ok

Unfortunately this is useless to me, as I have no time to learn how to add my own programs or customize my own firmware. The problem is that this official github hasn't been touched in over 3 years. For general users and those who could add to it just see an empty and bare repo and get discouraged. We shouldn't have to get source code from 3rd parties, it should be here. It seems the device has been given up on. What could have been something great has been left in the dust and forgotten. It honestly blows my mind that nobody shows interest, its just folks like us who don't have time or don't know how to code.

hramrach, 2024-01-14:
They even published a video on how to add your own program but not the source needed to do it.

Pushed the available 0.8.30 sources into a fork repo. Without the change history the usefulness is limited but it can be used as a starting point.

https://hackaday.io/project/159811-esp32-wiphone/log/195409-add-a-custom-button-function