ESP32-WiPhone/wiphone-firmware #47: It's been 4 years, Just release the source already

theSoberSobber, 2024-02-05:
It's been a dissapointment.

otilrac, 2024-10-14:
https://wiphone.io/static/releases/firmware/WiPhone-0.8.30.zip

brianmay, 2024-10-14:
That source code is old, 0.8.32rc2 is the latest version, I think.

So anybody who wanted to resume development would probably have to redo fixing bugs that were already resolved :-(

brianmay, 2024-10-14:
Supposedly this has 0.8.32 firmware: https://github.com/jonrick/wiphone - is that the same as 0.8.32rc2?

Linked from here: https://www.reddit.com/r/wiphone/comments/199jrxb/0832_firmware/

And there is a "hacker_edition" branch - what ever that is: https://www.reddit.com/r/wiphone/comments/199mtd7/hacker_edition_firmware/

Seems very dodgy. Do we know this is genuine? Who is Jon Rick, and what is his connection to wiphone?

I think they have lost interest. Fair enough. It is a big project. But the least they could do is publish the source from an official repository, maybe even including the history of changes.

otilrac, 2024-10-15:
I received the wiphone a few days ago, they say they are sent with the latest firmware the one I got was in 0.8.31, and the sources that I was given is 0.8 .30. so I compiled the 30 to put it on I used the 31 a little but I noticed that the 30 has more bugs the sources 32 that you showed me I have the impression that it is not good because for example the SIP part is missing, I am only going to use it for that and a couple of other little things, I I don't understand why they didn't give me the sources for 32 knowing that it's the last one, I'm going to ask but they take extremely long to respond in fact I think they don't answer the slightest question

rouing, 2024-12-04:
I got one of these second hand and now I know why.

brianmay, 2024-12-04:
It does seem like https://github.com/jonrick/wiphone is the best hope.